ika-rwth-aachen/mqtt_client #9: [Feature Request] sensor_msgs/JointState

Pedroacsilva, 2022-11-10:
Hello,

First of all, I'd like to thank you for publishing this, it's been very useful.

I'm interested in transmitting ROS messages typed `sensor_msgs/JointStates` to a broker outside of the ROS environment. As is, my MQTT subscriber, outside of the ROS environments, receives mostly gibberish. I presume this is because primitives messages do not support messages typed `sensor_msgs`. Are there any plans to support these?

lreiher, 2022-11-14:
Hi, happy that you are using the *mqtt_client*!

You are right, the primitive message support only covers the primitives (int, float, bool, ...) from `std_msgs`. What is happening in your case is that the sending *mqtt_client* serializes the `sensor_msgs/JointStates` message, but your receiving client apparently has no way of deserializing the incoming data.

I'm wondering how the target state would look like to you? Receiving a stringified JSON of the message?

Pedroacsilva, 2022-11-14:
Hello, thank you for the quick reply!

Yes, my goal is to receive messages in a JSON string, ideally. Right now, my MQTT client (Mosquitto) receives essentially gibberish and I presumed it was due to `/JointStates` not being a primitive message type (after reading some more, I don't believe that was the reason).

Is there support for mqtt_client to de/serialize ROS messages as JSON strings?

lreiher, 2022-11-15:
There is no support of such functionality as of now.

Unfortunately, it's not that easy to do from C++, without having access to proper type introspection at runtime. Python would be easy for this (and could also simply use [rosbridge_suite](http://wiki.ros.org/rosbridge_suite)).

[ros_msg_parser](https://github.com/facontidavide/ros_msg_parser) could potentially help to implement such a feature in our C++ node. We are already using a `topic_tools::ShapeShifter` to subscribe to generic message types. I would also like to try this out, but cannot yet guarantee that I will be getting to this anytime soon.

Pedroacsilva, 2022-11-15:
I'll look into these topics in order to implement this functionality. Thank you for your assistance and I'd like to apologize for not being very clear in my request -- clearly I didn't have a solid grasp on ROS-MQTT communications.

lreiher, 2022-11-15:
Nothing to apologize for! Feel free to open a pull request if you get this implemented in the mqtt_client!

lreiher, 2023-02-12:
This function could be used:

https://github.com/facontidavide/ros_msg_parser/blob/f874a6f98e0d1ed72e6d23f3671d26ed96a49986/include/ros_msg_parser/ros_parser.hpp#L160