ika-rwth-aachen/mqtt_client #12: Unable to bridge MQTT to ROS and ROS to MQTT

rajkumar2804, 2023-02-01:
I am trying to use mqtt_client on Ubuntu 20.04 with ROS noetic installed. Currently I have the following very simple .yaml config file and broker I am using is HiveMQ.
```
broker:
  host: "broker.mqttdashboard.com"
  port: 1883
bridge:
  ros2mqtt:
    - ros_topic: /ros_to_mqtt
      mqtt_topic: ros/mqtt
      advanced:
        ros:
          queue_size: 10 # [1] ROS subscriber queue size
        mqtt:
          qos: 2 # [0] MQTT QoS value
          retained: false # [false] whether to retain MQTT message
  mqtt2ros: #Only properly working for primitive datatype
    - mqtt_topic: mqtt/ros
      ros_topic: /mqtt_to_ros
      advanced:
        mqtt:
          qos: 2 # [0] MQTT QoS value
        ros:
          queue_size: 10 # [1] ROS publisher queue size
          latched: false # [false] whether to latch ROS message
      primitive: true
```
I have create few python scripts to test top communication. First problem I faced is if I don't set primitive true in the mqtt2ros and publish a message from a python script MQTT client on topic `mqtt/ros` I am getting any message on corresponding ROS topic. But if I set it true making sure I am sending a primitive data on topic `mqtt/ros` like string its get reflected on the ROS topic. Also the first message I send on MQTT topic is not received on corresponding ROS topic.

Next problem I faced is bridging of ROS topic to MQTT topic. here no primitive field is required to get the message reflected on MQTT topic from ROS topic. But I have a ros publisher 

node and a MQTT subscriber client script on the corresponding MQTT topic. If I start my MQTT sub client and then my  ROS publisher node so everytime first message published by my publisher get lost and not received on MQTT sub client. 

I am attaching a screenshot below 
[image]

lreiher, 2023-02-01:
Hi, thanks for using mqtt_client!

Regarding the primitive MQTT-to-ROS communication: it's expected that you have to set it to *primitive*. By default, the mqtt_client is waiting on information about the underlying ROS message type.

Otherwise, I'm not 100% sure I understood your issue correctly. The problem is that you are missing the first message or is there anything else?

It could also help if you posted your scripts that you tested this with along with the commands that you executed.

rajkumar2804, 2023-02-01:
Thanks, for you response
Regarding message the first ROS Message I figured it out, its because my ROS node is not initialised fully before publishing the data. But, I am not able to use the ROS-MQTT and MQTT-ROS communication for an arbitrary ROS message.

For example this bridge
```
  mqtt2ros:
    - mqtt_topic: mqtt/ros
      ros_topic: /mqtt_to_ros
      advanced:
        mqtt:
          qos: 2 # [0] MQTT QoS value
        ros:
          queue_size: 10 # [1] ROS publisher queue size
          latched: false # [false] whether to latch ROS message
```
I am using HiveMQ client dashboard to publish on  mqtt_topic  `/mqtt/ros` and the desired result should be message reflected on ROS topic `/mqtt_to_ros`. But its not the case. I am also going through the code and found a prefix  `kRosMsgTypeMqttTopicPrefix`  constant added before topic if not primitive is used does I have to publish the data on `kRosMsgTypeMqttTopicPrefix/mqtt/ros` and listen on `/kRosMsgTypeMqttTopicPrefix/mqtt_to_ros`.

lreiher, 2023-02-07:
As I have already mentioned, this is expected behavior. Please have a look at the "How it works" and "Primitve messages" sections again in the README. By default, the mqtt_client is waiting on information about the underlying ROS message type.

So if you are publishing primitive messages (e.g. strings) from something other than an mqtt_client (HiveMQ dashboard in your case), you have to set the receiving mqtt_client to primitive.

What kind of data are you publishing? What type of ROS message are you expecting?

lreiher, 2023-04-26:
Closing due to inactivity. Feel free to reopen if needed.

closed 2023-04-26 by lreiher