ika-rwth-aachen/mqtt_client #17: MQTT: No more messages can be buffered

aamirhatim, 2023-03-24:
I'm having an issue when publishing PointCloud2 messages where the mqtt publisher fails with the following message: `Publishing ROS message type information to MQTT topic 'jackal01/velodyne_points` failed: MQTT error [-12]: No more messages can be buffered`

I'm not too familiar with the MQTT library so I'm not sure what buffer this is referring to exactly but hoping to see if anyone has any suggestions on what I can look into to figure out what is going on. I tried adjusting the params for this package but no luck.

System specs:
- Jetson Nano
- Ubuntu 20.04
- ROS Noetic

Config file:
```
broker:
    host: 192.168.20.3
    port: 1883

client:
    id: client

bridge:
    ros2mqtt:
        - ros_topic: /jackal01/velodyne_points
          mqtt_topic: jackal01/velodyne_points
          primitive: false
          inject_timestamp: false
          advanced:
            ros:
                queue_size: 200
```

lreiher, 2023-03-25:
It looks like you are trying to send messages faster than the MQTT client library can deliver them. I guess this could either be due to your network connection being too slow, Jetson Nano being too slow, or perhaps you are not even connected to the broker continuously in the first place.

While you can play with the MQTT buffer size with the parameters `client/buffer/size` and `client/buffer/directory`, it probably makes more sense to evaluate your setup. I'm guessing that you are publishing point clouds at a constant rate, so increasing the buffer would only delay the issue.

At what rate are your point clouds being published? What happens if you set `client/buffer/size` and `client/buffer/directory`?

aamirhatim, 2023-03-27:
Messing with the buffer params didn't have any effect it seems. The messages themselves are coming at 10Hz but are pretty huge (the topic is publishing at ~100Mbps) so I think it may be a limitation from the Jetson - will need to investigate. I tried the exact same thing on a more powerful PC and there were no issues. 

Thanks for the quick response! I'll close this for now and re-visit if something else comes up.

closed 2023-03-27 by aamirhatim

JiaqiangZhang, 2023-05-01:
I meet the same issue but with different causes. My test is on 2 different laptops. I first set the `config` as follows:
```
mqtt_client:
  ros__parameters:
    broker:
      host: localhost
      port: 1883
    bridge:
      ros2mqtt:
        ros_topic: /test_topic/ping
        mqtt_topic: test_topic_mqtt/ros
      mqtt2ros:
        mqtt_topic: test_topic_mqtt/ros
        ros_topic: /test_topic/pong
```
One laptop send message with `ros2 topic pub /test_topic/ping std_msgs/msg/String "{data: \"Hello MQTT\"}"`, and the other one echo it with `ros2 topic echo /test_topic/pong`. Everything is fine in this situation, but when I change the `ros_topic` to the same one as this:
```
    bridge:
      ros2mqtt:
        ros_topic: /test_topic
        mqtt_topic: test_topic_mqtt/ros
      mqtt2ros:
        mqtt_topic: test_topic_mqtt/ros
        ros_topic: /test_topic
```
then I will keep getting this `warn` 
```
[mqtt_client-1] [WARN] [1682949299.441876554] [mqtt_client]: Publishing ROS message type information to MQTT topic 'test_topic_mqtt/ros' failed: MQTT error [-12]: No more messages can be buffered
```
The messages do transmit from one to the other, but the echo device receives more messages than the other pubs ( published 12 times, but received a full screen of message).

Could you please tell me why I cannot set `ros_topic` as the same? And is there a way to set `ros_topic` as the same?

lreiher, 2023-05-02:
Are you using the same config on both devices? If yes, you are entering a recursive loop:
- `ros2 topic pub` on laptop 1 publishes messages to `/test_topic` at 1Hz
- `mqtt_client` on laptop 1 forwards the message to the MQTT broker
- `mqtt_client` on laptop 2 receives the message and publishes it to ROS topic `/test_topic`
- `mqtt_client` on laptop 1 **also** receives the messages and publishes it to ROS topic `/test_topic`
- every message originally published by `ros2 topic pub` is now also received again on the same topic, thereby recursively increasing the publishing rate exponentially

Solution: configure either `ros2mqtt` or `mqtt2ros`, not both of them at the same time.

JiaqiangZhang, 2023-05-02:
Thanks for your reply. Solved this by configuring `ros2mqtt` and `mqtt2ros` separately.