ika-rwth-aachen/mqtt_client #18: How can I publish same ROS topic on multiple MQTT messages?

Lakshay011, 2023-04-20:
We attempted to publish a ROS topic message on multiple MQTT topics, but only one of the topics received the message. Is there a way to publish a single ROS message to multiple MQTT topics?

Config File:
> broker:
>     host: localhost
>     port: 1883
> bridge:
>     ros2mqtt:
>         -ros_topic: /agv/scan
>          mqtt_topic: agv_mqtt/scan/ros1
>          -ros_topic: /agv/scan
>          mqtt_topic: agv_mqtt/scan/ros2
>         -ros_topic: /agv/scan
>          mqtt_topic: agv_mqtt/scan/primitive/ros1
>          primitive: true

lreiher, 2023-04-26:
Hi, sorry for the delayed response.

What you are reporting is actually expected behavior. We are storing the ROS subscribers in a map structure, such that we can only store one ros2mqtt connection per ROS topic ([see this line of code](https://github.com/ika-rwth-aachen/mqtt_client/blob/main/include/mqtt_client/MqttClient.h#L433)).

I guess we could change this behavior or at least issue a warning, but first, could you please describe why you would want to publish the same data to multiple topics?

Lakshay011, 2023-05-10:
Apologies for the delay in getting back to you.

We currently have a situation where we need to transmit Lidar data to both a ROS-based environment and a web-based environment for visualization. The ROS-based environment is capable of decoding non-primitive data, whereas the web-based environment lacks a decoding library that can convert this data into a ROS object. Therefore, our aim is to send the Lidar data as primitive data to enable easy decoding.

If you are aware of any library that can assist us in decoding non-primitive messages into ROS object on the web side, please inform us.

lreiher, 2023-05-11:
For the web based interpretation, it would probably help a lot if we had https://github.com/ika-rwth-aachen/mqtt_client/issues/9, unfortunately that's an open issue at the moment.

For the other issue, that we cannot publish the same ROS topic to multiple MQTT topics, you would have two options at the moment:
- If you are familiar with C++ code, you could probably fix the current implementation pretty quickly to support multiple MQTT targets for each ROS topic, see my first answer.
- Otherwise, you could use [`topic_tools/relay`](https://github.com/ros-tooling/topic_tools#relay) to relay the ROS topic to a second topic that you would then both forward individually to MQTT.