ika-rwth-aachen/mqtt_client #19: Build failed for ROS2 Galacitc

JiaqiangZhang, 2023-04-26:
I am using galactic for my project. I tried install `mqtt_client` with `sudo apt install ros-galactic-mqtt-client`, and it gave me the error `E: Unable to locate package ros-galactic-mqtt-client`. 
Then I tried to build it from source. I run `colcon build --packages-up-to mqtt_client --cmake-args -DCMAKE_BUILD_TYPE=Release`,  first it gave me this error:
```
Starting >>> mqtt_client_interfaces
Finished <<< mqtt_client_interfaces [4.81s]                
Starting >>> mqtt_client
--- stderr: mqtt_client                         
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PahoMqttC_LIBRARY
    linked by target "mqtt_client" in directory /home/workspace/src/mqtt_client/mqtt_client
PahoMqttCpp_LIBRARY
    linked by target "mqtt_client" in directory /home/workspace/src/mqtt_client/mqtt_client

---
Failed   <<< mqtt_client [1.87s, exited with code 1]

Summary: 1 package finished [6.86s]
  1 package failed: mqtt_client
  1 package had stderr output: mqtt_client
```
After I build `paho.mqtt.c` and `paho.mqtt.cpp`, it gives me another error:
```
Starting >>> mqtt_client_interfaces
Finished <<< mqtt_client_interfaces [0.67s]                
Starting >>> mqtt_client
--- stderr: mqtt_client                             
/home/workspace/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp:35:10: fatal error: rcpputils/env.hpp: No such file or directory
   35 | #include <rcpputils/env.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mqtt_client.dir/build.make:63: CMakeFiles/mqtt_client.dir/src/MqttClient.ros2.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/mqtt_client.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< mqtt_client [0.31s, exited with code 2]

Summary: 1 package finished [1.16s]
  1 package failed: mqtt_client
  1 package had stderr output: mqtt_client
```

Could you please tell me how to solve this problem? Thanks!

lreiher, 2023-04-26:
First of all, we currently do not yet officially support ROS2. We have a [ROS2 branch](https://github.com/ika-rwth-aachen/mqtt_client/tree/dev/ros2) (that you checked out I guess) and an almost ready PR https://github.com/ika-rwth-aachen/mqtt_client/pull/16. The PR is currently blocked by https://github.com/ros/rosdistro/pull/36406. Until the PR in the rosdistro repository is resolved, the package will not be available for installation via `apt`.

Second, we are currently planning to only support ROS2 Humble, ROS2 Rolling, and onwards.

Nevertheless, I just tried to build this for Galactic anyway. The problem seems to be that the header file `env.hpp` of the `rcpputils` package was called `get_env.hpp` in Galactic. Modifying [the header include line](https://github.com/ika-rwth-aachen/mqtt_client/blob/dev/ros2/mqtt_client/src/MqttClient.ros2.cpp#L35) should allow you to successfully build our package.

Please let us know whether this has resolved the issue for you. We will consider whether to officially support Galactic as well.

JiaqiangZhang, 2023-04-26:
Thanks for your reply. Just had a quick try, after modifying `env.hpp` to `get_env.hpp`, `mqtt_client` build successfully, but comes with this warning:
```
Starting >>> mqtt_client_interfaces
Finished <<< mqtt_client_interfaces [0.73s]                
Starting >>> mqtt_client
--- stderr: mqtt_client                              
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp: In member function ‘virtual void mqtt_client::MqttClient::connected(const string&)’:
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp:840:47: warning: unused parameter ‘cause’ [-Wunused-parameter]
  840 | void MqttClient::connected(const std::string& cause) {
      |                            ~~~~~~~~~~~~~~~~~~~^~~~~
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp: In member function ‘virtual void mqtt_client::MqttClient::connection_lost(const string&)’:
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp:862:53: warning: unused parameter ‘cause’ [-Wunused-parameter]
  862 | void MqttClient::connection_lost(const std::string& cause) {
      |                                  ~~~~~~~~~~~~~~~~~~~^~~~~
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp: In member function ‘void mqtt_client::MqttClient::isConnectedService(mqtt_client_interfaces::srv::IsConnected_Request_<std::allocator<void> >::SharedPtr, mqtt_client_interfaces::srv::IsConnected_Response_<std::allocator<void> >::SharedPtr)’:
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp:878:64: warning: unused parameter ‘request’ [-Wunused-parameter]
  878 | nt_interfaces::srv::IsConnected::Request::SharedPtr request,
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~

/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp: In member function ‘virtual void mqtt_client::MqttClient::delivery_complete(mqtt::delivery_token_ptr)’:
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp:968:61: warning: unused parameter ‘token’ [-Wunused-parameter]
  968 | Client::delivery_complete(mqtt::delivery_token_ptr token) {}
      |                           ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~

/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp: In member function ‘virtual void mqtt_client::MqttClient::on_success(const mqtt::token&)’:
/home/workspace/py_subpub_ws/src/mqtt_client/mqtt_client/src/MqttClient.ros2.cpp:971:48: warning: unused parameter ‘token’ [-Wunused-parameter]
  971 | void MqttClient::on_success(const mqtt::token& token) {
      |                             ~~~~~~~~~~~~~~~~~~~^~~~~
---
Finished <<< mqtt_client [7.09s]

Summary: 2 packages finished [7.99s]
  1 package had stderr output: mqtt_client
```

Haven't test whether the node work properly. I will test this later today and give feedback.

JiaqiangZhang, 2023-04-26:
Also Checked [`rcpputils` package for `foxy`](https://github.com/ros2/rcpputils/blob/foxy/include/rcpputils/get_env.hpp). I think it's the same thing as `galactic`.

lreiher, 2023-04-27:
You can ignore those build warnings. Please let me know whether you can also successfully run the node with Galactic!

Supporting Foxy would involve more work I believe and we currently don't plan for it. The GenericSubscriber/GenericPublisher we use were only added to rclcpp with Galactic (https://github.com/ros2/rclcpp/pull/1452). I guess one could try to have a copy of their implementations in this repository, but it's likely some work to get it running.

JiaqiangZhang, 2023-04-27:
I tried the example of quick start, the node works fine. The messages are successfully sent and received in 2 terminals. Thanks for the instruction.

I might try to build this in Foxy later as well.

closed 2023-05-25 by lreiher

chrisjaxy, 2023-12-11:
> I tried the example of quick start, the node works fine. The messages are successfully sent and received in 2 terminals. Thanks for the instruction.
> 
> I might try to build this in Foxy later as well.

Have you successfully compiled and run it on foxy? Compared with Galacitc, what modifications need to be made to the foxy version?

lreiher, 2023-12-14:
Please see my comment above.

> Supporting Foxy would involve more work I believe and we currently don't plan for it. The GenericSubscriber/GenericPublisher we use were only added to rclcpp with Galactic [...]