ika-rwth-aachen/mqtt_client #26: terminate called after throwing an instance of 'ros::serialization::StreamOverrunException'   what():  Buffer Overrun

rahulelex, 2023-06-21:
### Logs
[ INFO] [1687354015.315784122]: Connecting to broker at 'tcp://192.168.200.54:1883' as 'clientid' ...
[ INFO] [1687354015.439690702]: Connected to broker at 'tcp://192.168.200.54:1883' as 'clientid'
[ INFO] [1687354015.450073212]: ROS publisher message type on topic '/jackal01/odometry/filtered' set to 'nav_msgs/Odometry'
[ INFO] [1687354015.451195642]: ROS publisher message type on topic '/tf' set to 'tf2_msgs/TFMessage'
[ INFO] [1687354015.452037397]: ROS publisher message type on topic '/tf_static' set to 'tf2_msgs/TFMessage'
[ INFO] [1687354015.452796619]: ROS publisher message type on topic '/jackal01/velodyne_packets' set to 'velodyne_msgs/VelodyneScan'
terminate called after throwing an instance of 'ros::serialization::StreamOverrunException'
  what():  Buffer Overrun
[mqtt_client-2] process has died [pid 34692, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone mqtt_client/MqttClient __name:=mqtt_client __log:=/home/ubuntu/.ros/log/48f05756-1037-11ee-9f81-0242ac110002/mqtt_client-2.log].
log file: /home/ubuntu/.ros/log/48f05756-1037-11ee-9f81-0242ac110002/mqtt_client-2*.log

### I also tried to add buffer size in configuration file.
client:
  id: "clientid"
  buffer:
    size: 10000
    directory: "/home/ubuntu"

### How can i increase the buffer size?

lreiher, 2023-06-21:
I don't think it has anything to do with the buffer size that you can configure. It's seems to crash during serialization of one of your messages.

Could you first try to narrow it down to a specific one of your topics by disabling all but one? Perhaps it also crashes with all topics. Please also post your entire config file.

I can also see that you are using ROS Melodic. Note that we do not officially support ROS Melodic (and it just went out-of-life) and frankly, I have also never tested it on Melodic. So perhaps you should consider upgrading to Noetic?

rahulelex, 2023-06-22:
I encountered an error message stating "Buffer overrun" while working with the MQTT package. This occurred after modifying the package by incorporating the neotic library. Previously, the package was functioning properly. The error emerged when I scaled up the data and added additional topics.

## params.yaml
broker:
  host: 192.168.200.54
  port: 1883
client:
  id: "clientid"
  buffer:
    size: 10000
    directory: "/home/ubuntu"
bridge:
  ros2mqtt:
    - ros_topic: /jackal01/move_base_simple/goal
      mqtt_topic: mqtt/move_base_simple/goal
    # - ros_topic: /floor_map/map
    #   mqtt_topic: mqtt/floor_map/map
    #   advanced:
    #     mqtt:
    #       retained: true
    # - ros_topic: /floor_map/map_metadata
    #   mqtt_topic: mqtt/floor_map/map_metadata
    #   # inject_timestamp: true
    #   advanced:
    #     mqtt:
    #       retained: true
    - ros_topic: /jackal01/scan
      mqtt_topic: mqtt/jackal01/scan
      # inject_timestamp: true
    - ros_topic: /jackal01/velodyne_points
      mqtt_topic: mqtt/jackal01/velodyne_points
      # inject_timestamp: true
    - ros_topic: /tf
      mqtt_topic: mqtt/jackal01/tf_from_amcl
      # inject_timestamp: true
  mqtt2ros:
    - mqtt_topic: mqtt/jackal01/velodyne_packets
      ros_topic: /jackal01/velodyne_packets
    - mqtt_topic: mqtt/jackal01/odometry_filtered
      ros_topic: /jackal01/odometry/filtered
    - mqtt_topic: mqtt/jackal01/tf
      ros_topic: /tf
    - mqtt_topic: mqtt/jackal01/tf_static
      ros_topic: /tf_static
      advanced:
        ros:
          latched: true

lreiher, 2023-06-22:
> This occurred after modifying the package by incorporating the neotic library

What is this supposed to mean?

The log from your first post suggests that you are using ROS Melodic?

`[mqtt_client-2] process has died [pid 34692, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet standalone`

rahulelex, 2023-06-23:
After investigating further, I have discovered that the error occurred due to modifications made to the package. I hereby declare the issue as resolved and will proceed to close it.

closed 2023-06-23 by lreiher