ika-rwth-aachen/mqtt_client #33: how to detect latency

hankge, 2023-09-27:
Hi all,
Thank you for your work. I want to use one big data(10MB) to test the transmit rate. I would like to know how to detect and output the latency. Actually I found you guys write some codes to get latency in mqttclient.cpp, but I don't know how to use it.

Thank you in advance!

hankge, 2023-09-27:
There is no ros topic like this:
/<mqtt_client_name>/latencies/<mqtt2ros/ros_topic>/data

lreiher, 2023-09-27:
Would you mind sharing your config file? Did you set the `inject_timestamp` parameter?

hankge, 2023-09-27:
I did not set that parameter. Here is my config.

```
mqtt_client:
  ros__parameters:
    broker:
      host: "broker.mqttdashboard.com"
      port: 1883
    bridge:
      ros2mqtt:
        ros_topics: 
          - /Robot1/image
        /Robot1/map:
          mqtt_topic: Robot1_map/map
      mqtt2ros:
        mqtt_topics: 
          - Robot1_map/map
        Robot1_map/map:
          ros_topic: Robot1_map/image
```

lreiher, 2023-09-27:
From the README:

> In order to inject the current timestamp into outgoing MQTT messages, the parameter `inject_timestamp` has to be set for the corresponding `bridge/ros2mqtt` entry. The receiving mqtt_client will then automatically publish the measured latency in seconds as a ROS `std_msgs/Float64` message on topic `/<mqtt_client_name>/latencies/<mqtt2ros/ros_topic>`.

Adjust your config to this:

```yml
mqtt_client:
  ros__parameters:
    broker:
      host: "broker.mqttdashboard.com"
      port: 1883
    bridge:
      ros2mqtt:
        ros_topics: 
          - /Robot1/image
        /Robot1/map: # <=== shouldn't this be /Robot1/image ?
          mqtt_topic: Robot1_map/map
          inject_timestamp: true # <===
      mqtt2ros:
        mqtt_topics: 
          - Robot1_map/map
        Robot1_map/map:
          ros_topic: Robot1_map/image
```

hankge, 2023-09-27:
I am using default yaml adding inject_timestamp parameter to test. There is no latency topic.
[image]

hankge, 2023-09-27:
I am not sure if this is the latency.
[image]

lreiher, 2023-09-27:
The `rosnode ping` is completely different.

Again, can you please post the exact config you are using such that I can try to reproduce the issue? Also, are you working on a single machine with a single `mqtt_client` or with multiple ones? If multiple, please post the config of all clients.

hankge, 2023-09-27:
This is what I am using now. I try it on ROS-noetic. 
```
broker:
  host: localhost
  port: 1883
bridge:
  ros2mqtt:
    - ros_topic: /ping/ros
      mqtt_topic: pingpong/ros
    - ros_topic: /ping/primitive
      mqtt_topic: pingpong/primitive
      primitive: true
      inject_timestamp: true
  mqtt2ros:
    - mqtt_topic: pingpong/ros
      ros_topic: /pong/ros
    - mqtt_topic: pingpong/primitive
      ros_topic: /pong/primitive
      primitive: true
``` 
For now, I only use one machine with one client. I will connect with multiple mechines in the future.
My lab system is ROS2, that’s why I give you the prior config.

lreiher, 2023-09-27:
Also from the README:

> Note that this functionality is only available for non-primitive messages (see [Primitive Messages](https://github.com/ika-rwth-aachen/mqtt_client/issues/33#primitive-messages))

hankge, 2023-09-27:
Okay, so if I want to use this functionality to test the latency, I need to transmit a type that are not primitive message. Right?
Can you recommend a dummy file that can easily get the latency?

lreiher, 2023-09-28:
I have just successfully tested this config with ROS2 Humble:

```yml
mqtt_client:
  ros__parameters:
    broker:
      host: localhost
      port: 1883
    bridge:
      ros2mqtt:
        ros_topics: 
          - /ping/ros
        /ping/ros:
          mqtt_topic: pingpong/ros
          inject_timestamp: true
      mqtt2ros:
        mqtt_topics: 
          - pingpong/ros
        pingpong/ros:
          ros_topic: /pong/ros
```

You can also transmit primitive types, the only thing you cannot do is enable the `primitive` flag in the config.

Test it by publishing a string:

```
ros2 topic pub -r 1 /ping/ros std_msgs/msg/String "{data: 'Hello World'}"
```

hankge, 2023-09-29:
Thank you so much @lreiher . I got the latency on ROS2 too. 
But I still confused about the latency topic. Based on your code ` ros::Duration latency = now - stamp;`, the latency is computed using arrrival time minus deserialized stamp. My question is do this functionality contains all the timestamps from publish to receive, including serialize and deserialize process.
BTW, Can ROS-noetic also do the latency check step? Because I tried to disable primitive, but there still not publish the latency topic.

lreiher, 2023-09-29:
The message stamp is taken [here](https://github.com/ika-rwth-aachen/mqtt_client/blob/207a618a75e6f56bff2922fc279cd9c71b8eacfb/mqtt_client/src/MqttClient.ros2.cpp#L660) and then injected into the already serialized message payload. What follows it the publication of the serialized ROS message via a generic publisher.

ROS Noetic should also work, I have also tried that out a couple of days ago. Note that the config file is a little different between ROS and ROS 2.

hankge, 2023-09-29:
Okay, so the latency does include the timestamps from publishing to receiving, including serializing because you get the time before serialize. Thank you so much. 

Actually this config is what I am using for ROS-noetic.
```
broker:
  host: localhost
  port: 1883
bridge:
  ros2mqtt:
    - ros_topic: /ping/ros
      mqtt_topic: pingpong/ros
    - ros_topic: /ping/primitive
      mqtt_topic: pingpong/primitive
      primitive: false
      inject_timestamp: true
  mqtt2ros:
    - mqtt_topic: pingpong/ros
      ros_topic: /pong/ros
    - mqtt_topic: pingpong/primitive
      ros_topic: /pong/primitive
      primitive: false
```

lreiher, 2023-10-04:
I have just checked the ROS 1 version again.

While I can indeed see latency values being published, I have to admit that there seems to be an issue with the latency publishing feature in the ROS 1 version. The published values seem random and after a short while the client crashes with the following error. I am going to investigate this further and hopefully publish a fix soon. The ROS 2 version was working for me though.

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
```

lreiher, 2023-10-04:
The ROS1 bug should be solved with https://github.com/ika-rwth-aachen/mqtt_client/pull/36.

hankge, 2023-10-05:
Thank you @lreiher. That works for me! :)

closed 2023-10-05 by lreiher