ika-rwth-aachen/mqtt_client #38: publish from MQTT to ROS

cboostkmokhtar, 2023-10-09:
I am not able to publish from mqtt to ros2

below is the configuration
`mqtt_client:
  ros__parameters:
    broker:
      host: 172.17.0.1
      port: 1883
    bridge:
      ros2mqtt:                     # Object specifying which ROS topics to map to which MQTT topics
        ros_topics:                 # Array specifying which ROS topics to bridge
          - /ping/ros               # The ROS topic that should be bridged, corresponds to the sub-object in the YAML
        /ping/ros:  
          mqtt_topic: pingpong/ros  # MQTT topic on which the corresponding ROS messages are sent to the broker
      mqtt2ros:                     # Object specifying which MQTT topics to map to which ROS topics
        mqtt_topics:                # Array specifying which ROS topics to bridge
          - pingpong/ros            # The MQTT topic that should be bridged, corresponds to the sub-object in the YAML
        pingpong/ros:
          ros_topic: /pong/ros      # ROS topic on which corresponding MQTT messages are published`

MQTT publisher on topic 

`topic = "pingpong/ros"`

if I do 'ros2 topic list' I would expect to see '/pong/ros' but I am not seeing it.

What do I miss?
By the way I can publish from ros2 to mqtt with ros2 topic `/ping/ros` and mqtt topic `pingpong/ros`

Cheers

lreiher, 2023-10-09:
What type of data are you publishing, how does the payload look like?

Please note that the `mqtt_client` is primarily intended to bridge ROS topics via MQTT, i.e., to bridge ROS messages. If you are sending an MQTT payload with another MQTT client (i.e., not the `mqtt_client` ROS package), the receiving `mqtt_client` cannot deserialize the payload to a ROS message.

We do, however, support such exchanges for primitive message types. Please read through the dedicated [README section](https://github.com/ika-rwth-aachen/mqtt_client#primitive-messages) and then try to set `primitive: true` for the `mqtt2ros` connection in your config file.

cboostkmokhtar, 2023-10-10:
Thank you. It works

I added `primitive: true` to my configuration file

To answer your question, I am publishing a primitive data type, string

closed 2023-10-10 by lreiher