ika-rwth-aachen/mqtt_client #39: Error launching package on Raspberry Pi

prbelarmino, 2023-11-02:
Hello,

I'm trying to use the package in a Raspberry Pi 3B+ with Raspbian OS and ROS noetic. When I launched the package I got the following messages:

[ WARN] [1550139515.070712783]: Parameter 'broker/tls/enabled' not set, defaulting to '0'
[ WARN] [1550139515.084676755]: Parameter 'client/id' not set, defaulting to ''
[ WARN] [1550139515.085581642]: Client buffer can not be enabled when client ID is empty
[ WARN] [1550139515.092249230]: Parameter 'client/clean_session' not set, defaulting to '1'
[ WARN] [1550139515.097297460]: Parameter 'client/keep_alive_interval' not set, defaulting to
'60.000000'
[ WARN] [1550139515.102260794]: Parameter 'client/max_inflight' not set, defaulting to '65535'
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /opt/ros/noetic/lib//libmqtt_client.so: undefined
symbol: _ZNSt10filesystem7__cxx114path14_M_split_cmptsEv
[mqtt_client-2] process has died [pid 2174, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet
standalone mqtt_client/MqttClient __name:=mqtt_client __log:=/home/pi/.ros/log/dcd419a6-3041-11e9-
93a2-0013252229ad/mqtt_client-2.log].
log file: /home/pi/.ros/log/dcd419a6-3041-11e9-93a2-0013252229ad/mqtt_client-2*.log

I believe the problem is during the compilation a flag/symbol was not set properly, but I don't know how I can fix it.

lreiher, 2023-11-02:
Have you built the package yourself or have you installed it via a package manager?

prbelarmino, 2023-11-03:
I did it by myself. Since the Raspberry Pi has a limited computing capacity. I installed the package on the Raspberry Pi in the following steps:
1- rosinstall_generator to generate the list of the package dependencies.
2- wstool to fetch all the remote repos specified from the dependencies.
3- catkin_make_isolated

lreiher, 2023-11-03:
The error message `undefined
symbol: _ZNSt10filesystem7__cxx114path14_M_split_cmptsEv` indicates that the issue is related to `std::filesystem` that is used in the implementation.

As far as I am aware, `std::filesystem` is only available starting with C++17. Note that the `CMakeLists.txt` is configured to build the `mqtt_client` with C++17. However, the `cxx114` in the error message sounds like perhaps you only have C++14 installed.

Have you tried building from source the way it is described in the [README](https://github.com/ika-rwth-aachen/mqtt_client#installation) (`rosdep`, `catkin build`)?

Alternatively, have you tried installing it from the package manager, i.e., `sudo apt-get install ros-noetic-mqtt-client`.

paulo-belarmino, 2023-11-20:
Hello, thank you for your reply. Based on what you wrote, I tried different approaches. First, I installed the package from the package manager, but I encountered the same error as when I installed it from the source. Then, I focused on the C++ version. I realized that my OS (Debian Buster) has GCC 8 by default, and Ubuntu 20, where I could install the package without any issues, has GCC 9. I attempted to upgrade the compiler, but my system got corrupted in the process.

Next, I tried to force GCC 8 to use C++17 standards. Finally, I found that, in addition to "add_compile_options(-std=c++17)," I also needed to add "link_libraries("$<$<AND:$<CXX_COMPILER_ID:GNU>,$<VERSION_LESS:$<CXX_COMPILER_VERSION>,9.0>>:-lstdc++fs>")" in the CMakeLists.txt file. After making these changes, I compiled again from the source using catkin_make_isolated, and finally, I could launch the package without any errors.

prbelarmino, 2023-11-20:
Hello, thank you for your reply. Based on what you wrote, I tried different approaches. First, I installed the package from the package manager, but I encountered the same error as when I installed it from the source. Then, I focused on the C++ version. I realized that my OS (Debian Buster) has GCC 8 by default, and Ubuntu 20, where I could install the package without any issues, has GCC 9. I attempted to upgrade the compiler, but my system got corrupted in the process.

Next, I tried to force GCC 8 to use C++17 standards. Finally, I found that, in addition to "add_compile_options(-std=c++17)," I also needed to add "link_libraries("$<$<AND:$<CXX_COMPILER_ID:GNU>,$<VERSION_LESS:$<CXX_COMPILER_VERSION>,9.0>>:-lstdc++fs>")" in the CMakeLists.txt file. After making these changes, I compiled again from the source using catkin_make_isolated, and finally, I could launch the package without any errors.

lreiher, 2023-11-20:
Good to hear! Feel free for posting a pull request to fix the issue!

PR #46 (2024-10-28): Update CMakeLists file to run on a raspbian buster system
Fixes #39
closed 2024-10-28 by lreiher