ika-rwth-aachen/mqtt_client #47: mqtt_client fails to start on Ubuntu 20.04 with ROS Noetic because of shared object error

nathanael-naeri, 2024-01-08:
Newbie here, I am following the instructions in the README file on Ubuntu 20.04 and ROS 1 Noetic, but running `roslaunch mqtt_client standalone.launch` fails immediately.

```
$ sudo apt install ros-noetic-mqtt-client
```

This command installed the following packages from packages.ros.org: ros-noetic-mqtt-client ros-noetic-mqtt-client-interfaces ros-noetic-paho-mqtt-c ros-noetic-paho-mqtt-cpp ros-noetic-rosfmt.

Then trying to run mqtt_client as per the README file:

```
$ roslaunch mqtt_client standalone.launch 
... logging to /home/nael/.ros/log/e5a750ae-ae3e-11ee-b43d-b519d361bafe/roslaunch-sandray-19552.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sandray:36191/

SUMMARY
========

PARAMETERS
 * /mqtt_client/bridge/mqtt2ros: [{'mqtt_topic': '...
 * /mqtt_client/bridge/ros2mqtt: [{'ros_topic': '/...
 * /mqtt_client/broker/host: localhost
 * /mqtt_client/broker/port: 1883
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    mqtt_client (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [19560]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e5a750ae-ae3e-11ee-b43d-b519d361bafe
process[rosout-1]: started with pid [19570]
started core service [/rosout]
process[mqtt_client-2]: started with pid [19577]
type is mqtt_client/MqttClient
[ERROR] [1704729566.420390979]: Failed to load nodelet [/mqtt_client] of type [mqtt_client/MqttClient] even after refreshing the cache: Failed to load library /opt/ros/noetic/lib//libmqtt_client.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libpaho-mqttpp3.so.1: cannot open shared object file: No such file or directory)
[ERROR] [1704729566.421195782]: The error before refreshing the cache was: Failed to load library /opt/ros/noetic/lib//libmqtt_client.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libpaho-mqttpp3.so.1: cannot open shared object file: No such file or directory)
[mqtt_client-2] process has died [pid 19577, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet standalone mqtt_client/MqttClient __name:=mqtt_client __log:=/home/nael/.ros/log/e5a750ae-ae3e-11ee-b43d-b519d361bafe/mqtt_client-2.log].
log file: /home/nael/.ros/log/e5a750ae-ae3e-11ee-b43d-b519d361bafe/mqtt_client-2*.log
```

Am I doing anything wrong? This is a standard installation of ROS 1 Noetic from ros.org, installed following their instructions.

Notes:

* The library `/opt/ros/noetic/lib//libmqtt_client.so` that reportedly fails to be loaded is present at that path, but note the double `/` in this string
* The file `libpaho-mqttpp3.so.1` that reportedly doesn't exist is at the following path: `/opt/ros/noetic/lib/x86_64-linux-gnu/libpaho-mqttpp3.so.1` and apparently it belongs to the APT package ros-noetic-paho-mqtt-cpp that was installed as a dependency of ros-noetic-mqtt-client
* My `ROS_PACKAGE_PATH` is standard and defined by `/opt/ros/noetic/setup.bash` to `/opt/ros/noetic/share` - should it be something else?
  ```
  $ printenv | grep ROS
  ROS_VERSION=1
  ROS_PYTHON_VERSION=3
  ROS_PACKAGE_PATH=/opt/ros/noetic/share
  ROSLISP_PACKAGE_DIRECTORIES=
  ROS_ETC_DIR=/opt/ros/noetic/etc/ros
  ROS_MASTER_URI=http://localhost:11311
  ROS_ROOT=/opt/ros/noetic/share/ros
  ROS_DISTRO=noetic
  ```
* Listing the dependencies of mqtt_client appears to show that some dependencies are missing or not found - why is that?
  ```
  $ rospack find mqtt_client
  /opt/ros/noetic/share/mqtt_client
  $ rospack depends1 mqtt_client
  [rospack] Error: package 'mqtt_client_interfaces' depends on non-existent package 'rosidl_default_runtime' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
  [rospack] Error: package 'mqtt_client' depends on non-existent package 'rclcpp' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
  [rospack] Error: package 'mqtt_client' depends on non-existent package 'rclcpp_components' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
  [rospack] Error: package 'mqtt_client' depends on non-existent package 'rcpputils' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
  [rospack] Error: package 'mqtt_client' depends on non-existent package 'paho-mqtt-cpp' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
  mqtt_client_interfaces
  ros_environment
  std_msgs
  nodelet
  roscpp
  rosfmt
  topic_tools
  ```
* Running `rosdep update` as suggested does not make a difference.

Thank you for your help!

nathanael-naeri, 2024-01-08:
I made this issue go away by changing
```
LD_LIBRARY_PATH=/opt/ros/noetic/lib
```
into
```
LD_LIBRARY_PATH=/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
```
I am not certain if that is a proper fix and if I should be altering that environment variable, but the Paho library seems to be in that additional directory:
```
$ ls -F -1 /opt/ros/noetic/lib/x86_64-linux-gnu
cmake/
libpaho-mqtt3a.so@
libpaho-mqtt3a.so.1@
libpaho-mqtt3a.so.1.3.13
libpaho-mqtt3as.so@
libpaho-mqtt3as.so.1@
libpaho-mqtt3as.so.1.3.13
libpaho-mqtt3c.so@
libpaho-mqtt3c.so.1@
libpaho-mqtt3c.so.1.3.13
libpaho-mqtt3cs.so@
libpaho-mqtt3cs.so.1@
libpaho-mqtt3cs.so.1.3.13
libpaho-mqttpp3.so@
libpaho-mqttpp3.so.1@
libpaho-mqttpp3.so.1.3.1
```

lreiher, 2024-01-08:
My guess is that re-sourcing ROS after installation should fix the issue.

```
sudo apt install ros-noetic-mqtt-client
source /opt/ros/noetic/setup.bash
roslaunch mqtt_client standalone.launch
```

The errors that `rospack find` outputs are due to `mqtt_client` supporting ROS and ROS 2 at the same time.

nathanael-naeri, 2024-01-09:
Probably. I shut down my computer yesterday, so ROS was re-sourced by my `.bashrc` today, and I can see that `LD_LIBRARY_PATH` is now `/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu`.

Thanks! I'm closing this issue. Would suggest that the README file is updated with that `source` command.

closed 2024-01-09 by nathanael-naeri