ika-rwth-aachen/mqtt_client #49: connection to aws IoT core via mqtt

alexandroze, 2024-01-18:
Hi 
Amazing project!
I have this locally rostopic '/robot1/battery_life' that I want to stream via mqtt to aws iot core. I was unable to do so when I modified the params.aws.yaml file; when I run the file, it stated:
[ WARN] [1705567736.240799367]: Parameter 'client/buffer/size' not set, defaulting to '0'
[ WARN] [1705567736.242808790]: Parameter 'client/buffer/directory' not set, defaulting to 'buffer'
[ WARN] [1705567736.245092743]: Parameter 'client/max_inflight' not set, defaulting to '65535'
[ INFO] [1705567736.247963741]: Bridging primitive ROS topic '/robot1/battery_life' to MQTT topic '/robot1/battery_life/primitive' 
[ INFO] [1705567736.248025214]: Bridging MQTT topic '/robot1/battery_life/primitive' to primitive ROS topic '/robot1/battery_life/mqtt'
[ INFO] [1705567736.250095406]: Connecting to broker at 'ssl://.................iot.ats.amazonaws.com:8883' as 'user' ...
[ERROR] [1705567736.401100272]: Connection to broker failed (return code -1), will automatically retry...
[ WARN] [1705567736.507493309]: Publishing ROS message type information to MQTT topic '/robot1/battery_life/primitive' failed: MQTT error [-3]: Disconnected
[ WARN] [1705567736.541004797]: Publishing ROS message type information to MQTT topic '/robot1/battery_life//primitive' failed: MQTT error [-3]: Disconnected

more information regarding the rostopic:
$ rostopic info /robot1/battery_life
Type: std_msgs/Int32

$ rostopic echo /robot1/battery_life
data: 70
---
data: 70
---

params.aws.yaml config file
  ros2mqtt:
    - ros_topic: /robot1/battery_life
      mqtt_topic: robot1/battery_life/primitive
      primitive: true
  mqtt2ros:
    - mqtt_topic: robot1/battery_life/primitive
      ros_topic: /robot1/battery_life/mqtt
      primitive: true

how can i approach with this issue? appreciate :)

rLoka, 2024-01-18:
I have the same issue when connecting to mosquito broker that is not on the localhost. Connecting to localhost broker does not produce: `Connection to broker failed (return code -1), will automatically retry..`

I am using **2.2.0**

lreiher, 2024-01-18:
@alexandroze: Hard to debug without more information. Can I please see your entrie params config file?

@rLoka: Sounds more like a networking issue on your side than anything going wrong with the mqtt_client. But feel free to open a dedicated issue for your problem, also adding more details about configuration, network setup, etc..

alexandroze, 2024-01-19:
Hi Ireiher, attached for your reference, pls lmk your thoughts!
[aws param config file.txt](https://github.com/ika-rwth-aachen/mqtt_client/files/13983674/aws.param.config.file.txt)

lreiher, 2024-01-19:
To be honest, I have no experience with AWS MQTT brokers. I don't see anything super suspicious in your config though.

Perhaps @dmbuck32 can help out? He's the one who has contributed the AWS config in https://github.com/ika-rwth-aachen/mqtt_client/pull/28.

Another thing I would try is whether you are able to connect to the broker with any other (non-ROS) mqtt client, e.g., [`mosquitto_pub`](https://mosquitto.org/man/mosquitto_pub-1.html)

alexandroze, 2024-01-19:
Your detailed repo guidance has helped me, no issue connecting to mosquitto broker. Thank you so much and for introducing @dmbuck32.

hi @dmbuck32, need to tap on your expertise on this issue, appreciate your help.

babakc, 2024-01-22:
Hello from AWS! 

I was able to test this without issue. Just looking at the log you provided... "ssl://.................iot.ats.amazonaws.com:8883" which is not the correct formatting for the endpoint name, but it looks like you had the correct format in your params file. Just want to check if this was a miss spell when you masked your endpoint name?

If so, i'm assuming this is probably related to your policy in IoT core that's attached to the certificate you're using. Are you able to ensure the policy allows you to connect with the client id you provided? To rule out policy issues, i usually allow iot:* for resources * (but be sure to revert back once testing complete).

dmbuck32, 2024-01-22:
> Your detailed repo guidance has helped me, no issue connecting to mosquitto broker. Thank you so much and for introducing @dmbuck32.
> 
> hi @dmbuck32, need to tap on your expertise on this issue, appreciate your help.

Hi @alexandroze, I would tend to agree with @babakc that you might need to check the name of the AWS IoT endpoint or the IoT core policy. The policy should be active on the user that you use to connect to the IoT endpoint. Also make sure you don't have a policy that prevents multiple connections.

I've used the following command to get the IoT endpoint in the past:
```bash
aws iot describe-endpoint
```

babakc, 2024-01-23:
Ah yes, I forgot to mention, we recommend using the iot:Data-ATS endpoint type which contains the -ats suffix on the end of the hostname. To get that, use the cli:

`aws iot describe-endpoint --endpoint-type iot:Data-ATS`

Details are here: https://docs.aws.amazon.com/cli/latest/reference/iot/describe-endpoint.html

zeesee, 2024-01-24:
hi all, thanks for the help, managed to get it stream! it was the endpoint mistake :)

alexandroze, 2024-01-24:
solved it.. appreciate the help, many thanks!

closed 2024-01-24 by lreiher