ika-rwth-aachen/mqtt_client #59: Does not work with the example configuration

emoPointer, 2024-04-07:
Hello,

I am trying to use mqtt in my project, I follow your steps on wiki,installed mqtt via apt,and then i run 
'''
ros2 launch mqtt_client standalone.launch.ros2.xml
'''
All the configurations are default.
but it did not work  properly
error is below:
[image]

emoPointer, 2024-04-07:
Do you have to open a broker when use mqtt_client?
i used mqttx v1.9.10  to connect the ip of raspberrypi ,but it said 
    
Error: connect ECONNREFUSED 192.168.162.41:1883

then i changed ip to the ip of myself computer, it did not work as before
log is below:
[image]

Looking forward to your reply

emoPointer, 2024-04-07:
I can ping through my computer and Raspberry PI normally

bastilam, 2024-04-07:
The mqtt_client's [default ROS 2 configuration](https://github.com/ika-rwth-aachen/mqtt_client/blob/main/mqtt_client/config/params.ros2.yaml#L4) will try to connect to a broker on your localhost. If the broker is not running or if it is not reachable, you will get your initial error (`Connection to Broker failed`).

To start a broker on your localhost, [install Docker](https://docs.docker.com/engine/install/ubuntu/#install-using-the-repository) if you haven't already (including the [postinstall steps](https://docs.docker.com/engine/install/linux-postinstall/)), then run `docker run --rm --network host --name mosquitto eclipse-mosquitto`.

Then, you can start the mqtt_client, e.g., in a Docker container: `docker run --network host --rm ghcr.io/ika-rwth-aachen/mqtt_client:ros2`

You should then get a message `Connected to broker at 'tcp://localhost:1883'`

emoPointer, 2024-04-08:
thanks for your reply, i run it sucessufully with docker, but I feel that every time I have to open docker is taking up system resources. I am working on a Raspberry PI project and I want to reduce this unnecessary overhead. It is amazing to use docker to solve the problem in one fell fell because I have spent a long time on them

emoPointer, 2024-04-08:
I already know the reason for my failure.i should install mosquitto before run it.
can use command below:
''''
sudo apt-get install mosquitto
sudo apt-get install mosquitto-clients
''''

closed 2024-04-08 by emoPointer