ika-rwth-aachen/mqtt_client #62: mqtt_client Docker unable to connect to MQTT broker after startup

weyne-Jiang, 2024-05-09:
After I started mqtt_client container docker, the logs showed that several mqtt related parameters were not set, and I ended up unable to connect to mqtt broker. However, I have opened the emqx service locally in Ubuntu, and I can connect successfully through the browser. Q: What is the reason for this, and where are the MQTT parameters set for the mqtt_client container? thanks for your help.
this is log:

weyne@ubuntu:~$ docker run --rm ghcr.io/ika-rwth-aachen/mqtt_client:ros2
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-09-06-41-58-305609-513f020f5e52-1
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mqtt_client-1]: process started with pid [52]
[mqtt_client-1] [WARN] [1715236918.544429235] [mqtt_client]: Parameter 'broker.tls.enabled' not set, defaulting to '0'
[mqtt_client-1] [WARN] [1715236918.544538293] [mqtt_client]: Parameter 'client.id' not set, defaulting to ''
[mqtt_client-1] [WARN] [1715236918.544548242] [mqtt_client]: Client buffer can not be enabled when client ID is empty
[mqtt_client-1] [WARN] [1715236918.544556117] [mqtt_client]: Parameter 'client.clean_session' not set, defaulting to '1'
[mqtt_client-1] [WARN] [1715236918.544576185] [mqtt_client]: Parameter 'client.keep_alive_interval' not set, defaulting to '60.000000'
[mqtt_client-1] [WARN] [1715236918.544583539] [mqtt_client]: Parameter 'client.max_inflight' not set, defaulting to '65535'
[mqtt_client-1] [INFO] [1715236918.544604920] [mqtt_client]: Bridging ROS topic '/ping/ros' to MQTT topic 'pingpong/ros'
[mqtt_client-1] [INFO] [1715236918.544621061] [mqtt_client]: Bridging MQTT topic 'pingpong/ros' to ROS topic '/pong/ros'
[mqtt_client-1] [INFO] [1715236918.545433149] [mqtt_client]: Connecting to broker at 'tcp://localhost:1883' ...
[mqtt_client-1] [ERROR] [1715236918.746983785] [mqtt_client]: Connection to broker failed (return code -1), will automatically retry...
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[INFO] [mqtt_client-1]: sending signal 'SIGINT' to process[mqtt_client-1]
[mqtt_client-1] [INFO] [1715236920.810983658] [rclcpp]: signal_handler(signum=2)
[INFO] [mqtt_client-1]: process has finished cleanly [pid 52]

weyne@ubuntu:~$ systemctl status emqx
● emqx.service - emqx daemon
     Loaded: loaded (/lib/systemd/system/emqx.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-05-08 23:13:19 PDT; 28min ago
   Main PID: 995 (beam.smp)
      Tasks: 41 (limit: 4556)
     Memory: 252.7M
     CGroup: /system.slice/emqx.service
             ├─ 995 emqx -Bd -spp true -A 4 -IOt 4 -SDio 8 -C multi_time_warp -e 262144 -zdbbl 8192 -Q 1048576 -P 2097152 -- -root /usr/lib/emqx -bindir /usr/lib/emqx/erts-14.2.1/bin -progname /usr/bin/emq>
             ├─1889 erl_child_setup 1048576
             ├─1925 /usr/lib/emqx/lib/os_mon-2.9.1/priv/bin/memsup
             ├─1926 /usr/lib/emqx/lib/os_mon-2.9.1/priv/bin/cpu_sup
             ├─2150 /usr/lib/emqx/erts-14.2.1/bin/inet_gethost 4
             └─2151 /usr/lib/emqx/erts-14.2.1/bin/inet_gethost 4

May 08 23:13:19 ubuntu systemd[1]: Started emqx daemon.
May 08 23:13:23 ubuntu bash[995]: WARNING: Default (insecure) Erlang cookie is in use.
May 08 23:13:23 ubuntu bash[995]: WARNING: Configure node.cookie in /etc/emqx/emqx.conf or override from environment variable EMQX_NODE__COOKIE
May 08 23:13:23 ubuntu bash[995]: WARNING: NOTE: Use the same cookie for all nodes in the cluster.
May 08 23:13:27 ubuntu bash[995]: Listener tcp:default on 0.0.0.0:1883 started.
May 08 23:13:27 ubuntu bash[995]: Listener ssl:default on 0.0.0.0:8883 started.
May 08 23:13:27 ubuntu bash[995]: Listener ws:default on 0.0.0.0:8083 started.
May 08 23:13:27 ubuntu bash[995]: Listener wss:default on 0.0.0.0:8084 started.
May 08 23:13:28 ubuntu bash[995]: Listener http:dashboard on :18083 started.
May 08 23:13:28 ubuntu bash[995]: EMQX 5.6.1 is running now!

bastilam, 2024-05-09:
You could try giving the docker container in which the mqtt_client is running access to your host's network:

Instead of

`docker run --rm ghcr.io/ika-rwth-aachen/mqtt_client:ros2`

try

`docker run --network host --rm ghcr.io/ika-rwth-aachen/mqtt_client:ros2`.

In addition, you may first want to try the broker we describe in our readme.

So, before you start the mqtt_client, start the Mosquitto broker with

`docker run --rm --network host --name mosquitto eclipse-mosquitto`

You might need to stop the EMQX service temporarily before running the Mosquitto broker.

After you started the Mosquitto broker, run

`docker run --network host --rm ghcr.io/ika-rwth-aachen/mqtt_client:ros2`

as described above.

Let me know whether any of this helped!

lreiher, 2024-05-09:
To add some more explanation: I'm pretty sure that the issue is the following:
- the [default config](https://github.com/ika-rwth-aachen/mqtt_client/blob/main/mqtt_client/config/params.ros2.yaml) is configured to connect to a broker at `localhost:1883`
- when running the client inside a container, `localhost` refers to the container, not your host computer where your broker is apparently running
- starting the container with `--network host`, as @bastilam has suggested, will let the container use the host's network interfaces, I.e., `localhost` will refer to your host computer
- instead of sharing all network interfaces with the host, you could also configure the client to connect to `host.docker.internal` instead of `localhost`, which should resolve to the IP of the host from inside the client container
- you can use your own [config file](https://github.com/ika-rwth-aachen/mqtt_client/blob/main/mqtt_client/config/params.ros2.yaml) by mounting it into the container and the modifying the container launch command

```
docker run --rm -v $YOUR_CONFIG_FILE:/params.yml ghcr.io/ika-rwth-aachen/mqtt_client:ros2 ros2 launch mqtt_client standalone.launch.ros2.xml params_file:=/params.yml
```

weyne-Jiang, 2024-05-09:
Thank you very much, it's up and running now

closed 2024-05-10 by lreiher