ika-rwth-aachen/mqtt_client #71: Client could not be initialized: MQTT error [-2]: Persistence error

santiagoUnlitrobotics, 2024-09-11:
Hi, I'm trying to execute the following configuration.

```
import os
from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    CLIENT_ID = os.getenv('CLIENT_ID')
    ROBOT_ID = os.getenv('ROBOT_ID')
    
    node_parameters = {
        'broker': {
            'host': '...-ats.iot.eu-central-1.amazonaws.com',
            'port': 8883,
            'tls': {
                'enabled': True,
                'ca_certificate': '/robot/component_static_data/mqtt_certificates/JetsonOrin/root-CA.crt',
            },
        },
        'client': {
            'id': 'raya_monitor',
            'clean_session': False,
            'tls': {
                'enabled': True,
                'certificate': '/robot/component_static_data/mqtt_certificates/JetsonOrin/cert.pem',
                'key': '/robot/component_static_data/mqtt_certificates/JetsonOrin/private.key',
                'version': 3,
                'verify': True,
            },
            'buffer': {
                'size': 0,
                'directory': '/robot/component_static_data/buffer',
            },
        },
        'bridge': {
            'ros2mqtt': {
                'ros_topics': [
                    '/monitor/stats',
                    '/monitor/get_map/response',
                ],
                '/monitor/stats': {
                    'mqtt_topic': f'{CLIENT_ID}/robots/{ROBOT_ID}/monitor/stats',
                    'primitive': False,
                    'ros_type': 'raya_monitor_msgs/msg/Stats',
                },
                '/monitor/get_map/response': {
                    'mqtt_topic': f'{CLIENT_ID}/{ROBOT_ID}/monitor/get_map/response',
                    'primitive': False,
                    'ros_type': 'std_msgs/String',
                },
            },
            'mqtt2ros': {
                'mqtt_topics': [
                    f'{CLIENT_ID}/{ROBOT_ID}/monitor/get_map/request',
                ],
                f'{CLIENT_ID}/{ROBOT_ID}/monitor/get_map/request': {
                    'ros_topic': '/monitor/get_map/request',
                    'primitive': False,
                    'ros_type': 'std_msgs/String',
                }
            },
        },
    }

    
    return LaunchDescription([
        Node(
            package='raya_monitor',
            namespace='/raya/monitor',
            executable='raya_monitor_node',
            name='robot_monitor'
        ),
        Node(
            package='mqtt_client',
            namespace=f'/{CLIENT_ID}/robots/{ROBOT_ID}/monitor',
            executable='mqtt_client',
            name='mqtt_client',
            output='screen',
            parameters=[node_parameters]
        )
    ])
```

The output with this configuration is:
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-09-11-21-50-50-307954-Santiago-UR-3307170
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [raya_monitor_node-1]: process started with pid [3307231]
[INFO] [mqtt_client-2]: process started with pid [3307233]
[raya_monitor_node-1] [INFO] [1726091450.360234616] [raya.monitor.robot_monitor]: Raya Monitor initialized.
[raya_monitor_node-1] [INFO] [1726091450.360325517] [raya.monitor.robot_monitor]: Initializing ROS.
[raya_monitor_node-1] [INFO] [1726091450.360334163] [raya.monitor.robot_monitor]: NODE_RUNNING_AND_SERVING
[mqtt_client-2] [WARN] [1726091450.361462083] [Temple_University_Hospital.robots.SIMULATOR001.monitor.mqtt_client]: Parameter 'client.keep_alive_interval' not set, defaulting to '60.000000'
[mqtt_client-2] [WARN] [1726091450.361518960] [Temple_University_Hospital.robots.SIMULATOR001.monitor.mqtt_client]: Parameter 'client.max_inflight' not set, defaulting to '65535'
[mqtt_client-2] [INFO] [1726091450.361555028] [Temple_University_Hospital.robots.SIMULATOR001.monitor.mqtt_client]: Bridging ROS topic '/monitor/stats' to MQTT topic 'Temple_University_Hospital/robots/SIMULATOR001/monitor/stats' 
[mqtt_client-2] [INFO] [1726091450.361565748] [Temple_University_Hospital.robots.SIMULATOR001.monitor.mqtt_client]: Bridging ROS topic '/monitor/get_map/response' to MQTT topic 'Temple_University_Hospital/SIMULATOR001/monitor/get_map/response' 
[mqtt_client-2] [INFO] [1726091450.361588892] [Temple_University_Hospital.robots.SIMULATOR001.monitor.mqtt_client]: Bridging MQTT topic 'Temple_University_Hospital/SIMULATOR001/monitor/get_map/request' to ROS topic '/monitor/get_map/request'
[mqtt_client-2] [ERROR] [1726091450.362170405] [Temple_University_Hospital.robots.SIMULATOR001.monitor.mqtt_client]: Client could not be initialized: MQTT error [-2]: Persistence error
[ERROR] [mqtt_client-2]: process has died [pid 3307233, exit code 1, cmd '/opt/ros/humble/lib/mqtt_client/mqtt_client --ros-args -r __node:=mqtt_client -r __ns:=/Temple_University_Hospital/robots/SIMULATOR001/monitor --params-file /tmp/launch_params_lfsri8pj'].
```

What is causing this persistence Error??

santiagoUnlitrobotics, 2024-09-11:
I tested the connection to the broker with the primitive examples and it worked as expected so i know that the connection is not the problem.

lreiher, 2024-09-12:
Hard to debug and a little out of scope of the `mqtt_client` itself. Can you connect using other MQTT clients such as [`mosquitto_sub`](https://mosquitto.org/man/mosquitto_sub-1.html)? If yes, what does the command have to look like?

santiagoUnlitrobotics, 2024-09-20:
I close this, i read some other issues related to using non primitive types, in my case I changed to use the string type and format the information on JSON.

closed 2024-09-20 by santiagoUnlitrobotics