ika-rwth-aachen/mqtt_client #76: How to properly send twist message from non-ros to ros

leets0429, 2024-10-22:
I have a webapp (non-ros) sending twist message to the broker. Here is how I define the twist message in js

```
let twist = {
    linear: {
        x: 0.0,
        y: 0.0,
        z: 0.0
    },
    angular: {
        x: 0.0,
        y: 0.0,
        z: 0.0
    }
}
```

Here is my mqtt_client param
```
bridge:
  mqtt2ros:
    - mqtt_topic: cmd_vel
      ros_topic: /cmd_vel
```

The topic is converted into geometry_msgs/Twist, however I received weird value even though I'm sending twist with no speed. This is the value I receive using rostopic echo
```
---
linear: 
  x: 9.279857549485897e+242
  y: 2.2570320461507035e-76
  z: 4.5148394577894216e-144
angular: 
  x: 7.98459892965631e+159
  y: 1.164156688845003e-28
  z: 4.514839457752175e-144
---
```

I'm running ROS Noetic of Ubuntu 20.04

lreiher, 2024-10-24:
Hi, thanks for your interest in the mqtt_client!

The mqtt_client is primarily meant to bridge ROS messages from one ROS system to another ROS system via MQTT. What you are trying to do, however, is bridging messages from ROS to MQTT or from MQTT to ROS.

It's expected that this currently only works for *primitive* message types. For non-primitive message types such as `Twist`, the ROS messages are serialized to a binary payload that is then sent via MQTT. This binary payload is not directly human-readable and has to be deserialized by another mqtt_client to then publish as a ROS message on the receiver side.

In principle, you can always come up with a JSON representation of any ROS message, so why not send JSON equivalents of the ROS messages via MQTT? Although this would most likely increase the payload size, it could definitely be a very helpful feature for the mqtt_client, see https://github.com/ika-rwth-aachen/mqtt_client/issues/9!

As of today, this feature is not yet available on the main branch. There is, however, a pull request https://github.com/ika-rwth-aachen/mqtt_client/pull/51 that implements this feature. Please refer to the PR discussion on how to use it and why it hasn't been merged yet.

Hope this helps!

leets0429, 2024-10-25:
Alright, thank you very much for your reply

closed 2024-10-25 by leets0429