ika-rwth-aachen/mqtt_client #78: About how to use mqtt2ros publish more rostopic?

MA-xiaowen, 2024-11-20:
I was written this message to yaml file:
mqtt2ros:
    - mqtt_topic: pingpong/ros
      ros_topic: /pong/ros
    - mqtt_topic: pingpong/primitive
      ros_topic: /pong/primitive
    - mqtt_topic: Send/Navigation
      ros_topic: /Send/Navigation
But I only can received first and third topic message , I want know how to figure this error?

lreiher, 2024-11-20:
I would need more information to help here: full config file (on both sides: send and receive), how are you launching it, ...

MA-xiaowen, 2024-11-21:
can you give me your e-mail? I send all the projects to you.

MA-xiaowen, 2024-11-21:
And more, I write a python file to received the mqtt messages, I found it can receive it all . I debug this project found that I published  three message only two can arrive to "void MqttClient::message_arrived " function . I don't know why this things would happen .

MA-xiaowen, 2024-11-21:
Good news , I figure out why this things will happen , because I use the example yaml file , and the 'primitive' parameter was set in last line , I should set it as true in every mqtt_topic like this:
    - mqtt_topic: Send/Navigation
      ros_topic: /Send/Navigation
      primitive: true        # [false] whether to publish as primitive message (if coming from non-ROS MQTT client)
      advanced:
        mqtt:
          qos: 1          # [0] MQTT QoS value
        ros:
          queue_size: 10   # [1] ROS publisher queue size
so every mqtt_topic can transform to ros_topic.
And to the end , thank u for your excellence job, hope you happy everyday.

closed 2024-11-21 by lreiher